profezzorn/ProffieOSDocs #19: document all styles/functions/transitions

profezzorn, 2023-02-13:
Probably all we need is a page which lists all styles and then has links to the documentation that's in the source files.
Right now a complete list of all functions doesn't actually exist anywhere in a single place.

NoSloppy, 2024-07-03:
This a good start?
https://www.dropbox.com/scl/fo/xr2p1aexsunptbzqezepa/AP7MqtjquLYx9LtfGRyFb1w?rlkey=fk275wgy7an30j2xthkr8ev5k&dl=0

profezzorn, 2024-07-03:
Indeed, it is a good start.
I think we want one page per transition/style/function though. It will make it easy to use redirects for aliases and "X" functions. The most important part would be to have predictable URLs so that we can link to the pages from the style editor (and other tools)

Each directory (styles/functions/transitions) can have a generic index file to generate an index of all the functions, styles and transitions.

For people who still want to have a single page for browsing, the automatically generated "all the documentation on one page" should work I think.

NoSloppy, 2024-07-03:
wait... i knew we talked about this before....

https://github.com/profezzorn/ProffieOSDocs/pull/52

profezzorn, 2025-02-28:
Done.

closed 2025-02-28 by profezzorn