profezzorn/ProffieOSDocs #54: Board rotations not clearly defined

ryryog25, 2024-03-29:
On the POD "CONFIG_TOP" page, there is information about the `ORIENTATION_ROTATION` define, but while a brief example is given, a clear description of which direction is X, Y, and Z is not present.

https://pod.hubbe.net/config/the-config_top-section.html#orientationrotation

I'd think something like "The Y Axis is along the FETs and USB, the Z axis is perpendicular to the PCB plane, and the X axis runs between the FETs and USB" would help clarify.

I'd assume that the example description I gave is at least partially wrong (I don't know which axis are which directions) and/or could be better worded...

profezzorn, 2024-03-29:
I think your description is correct.
Pictures would be better I think.
PRs would be welcome.
Not a lot of people use this define since https://pod.hubbe.net/config/the-config_top-section.html#orientation-defines is enough for most people, so it hasn't been a high priority.

ryryog25, 2024-03-29:
Ok, I'd be happy to make a PR, but just wanted to double check those descriptions (can also come up with an image too), since it came to light due to someone else trying to use them and it not seeming to give the desired effect.

It was in the r/lightsabers discord, which I think you're in, so I'll [link the conversation here](https://discord.com/channels/665548882121457684/679392386320695338/1223354155775824093)